gphoto/libgphoto2 #1093: Sony RX100M7 remote video capture works with Segmentation fault

tvendelin, 2025-02-28:
**Describe the bug**
This successfully triggers video recording onto camera's SD card. The video is in high quality as expected. The video can also be stopped as shown. The commands, however, runs into a segfault:
```bash
# Start recording
>gphoto2 --set-config /main/actions/movie=1 
Segmentation fault
# Stop recording
>gphoto2 --set-config /main/actions/movie=0
Segmentation fault
```
$? is 139

**Name the camera**
Camera name as shown by gphoto2 --auto-detect or USB IDs:
```
> gphoto2 --auto-detect
Model                          Port                                            
----------------------------------------------------------
Sony DSC-RX100M7 (PC Control)  usb:001,028
```

**libgphoto2 and gphoto2 version**

```
>gphoto2 --version
gphoto2 2.5.28

Copyright (c) 2000-2021 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.28         cc, popt(m), exif, no cdk, no aa, jpeg, readline
libgphoto2      2.5.31         standard camlibs (SKIPPING lumix), cc, no ltdl, EXIF
libgphoto2_port 0.12.2         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, cc, no ltdl, EXIF, USB, serial without locking
```

[my-logfile.txt](https://github.com/user-attachments/files/19027849/my-logfile.txt)

**To Reproduce**
See the bug description.

msmeissn, 2025-02-28:
can you try to get a backtrace in gdb or so too please?

msmeissn, 2025-02-28:
fixed in committ bbb7dbed2b96fe41045c186fb2eff0035cde4224 , see https://github.com/gphoto/libgphoto2/issues/974
will be in next release

closed 2025-02-28 by msmeissn