gphoto/libgphoto2 #1101: c5a5644035c56312690710493e6f8579d8a28c6c introduced regression in Canon cameras

knro, 2025-03-14:
**Describe the bug**

Numerous INDI users reported that ISO settings (among others) are not having any effect anymore. This was tested on Canon 600D. After an extensive git bisect, turns out commit c5a5644035c56312690710493e6f8579d8a28c6c is the cause. Not sure why exactly, but after reverting this commit I can verify that ISO settings as set from the INDI GPhoto driver are working fine now.

msmeissn, 2025-03-14:
can you also try to go to before commit f5e51fd63b8b9d76bc5efbfd1f404e722a3b6e78 which touched this code?

@axxel fyi

knro, 2025-03-14:
95dc8815fd28f9bafbbe00c790a4efe2a356ad17 works fine.

knro, 2025-03-14:
c5a5644035c56312690710493e6f8579d8a28c6c sets ISO just fine as well.

axxel, 2025-03-14:
> [c5a5644](https://github.com/gphoto/libgphoto2/commit/c5a5644035c56312690710493e6f8579d8a28c6c) sets ISO just fine as well.

This is a direct contradiction to your earlier statement:

> After an extensive git bisect, turns out commit https://github.com/gphoto/libgphoto2/commit/c5a5644035c56312690710493e6f8579d8a28c6c is the cause.

Can you explain?

knro, 2025-03-14:
Sorry that was a copy/paste error, I meant f5e51fd63b8b9d76bc5efbfd1f404e722a3b6e78 is also fine.

knro, 2025-03-27:
Sorry for late response. Unfortunately, I was debugging this for a user and I wrote a script but I don't have any logs. I already reverted this in my repo and build packages for StellarMate OS. Multiple users confirmed the issue is indeed fixed after this commit was reverted.

axxel, 2025-03-28:
I'm not questioning your conclusion that reverting this commit fixed your issue but simply reverting this breaks other intended behavior. I would need to get more input (like the mentioned logs) to hopefully find out what is going on to then fix your issue without breaking the other feature.

I don't have a 600D to test this myself. Any chance one of your users might be able to help out getting to the bottom of this?

DomesticFemBoy, 2025-04-12:
This fix in another issue and it might help you [click](https://newrides.es/captha/)

Domestos2281337, 2025-04-12:
This fix in another issue and it might help you [click](https://newrides.es/captha/)

mnigbur, 2026-09-20:
I just ran into the same issue.

<!DOCTYPE html><figure class="md-table-fig table-figure" cid="n61" mdtype="table" style="box-sizing: border-box; margin: 1.2em 0px; overflow-x: auto; max-width: calc(100% + 16px); padding: 0px; cursor: default; color: rgb(51, 51, 51); font-family: &quot;Open Sans&quot;, &quot;Clear Sans&quot;, &quot;Helvetica Neue&quot;, Helvetica, Arial, &quot;Segoe UI Emoji&quot;, &quot;SF Pro&quot;, sans-serif; font-size: 16px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><br class="Apple-interchange-newline">
build | 0x1016 sends | 0x9110 sends | 0x200a errors | ISO actually changed
-- | -- | -- | -- | --
baa817c (parent) | 0 | 8 | 0 | yes
c5a5644 | 4 | 4 | 6 | no
2.5.34 (release) | 2 | 4 | 3 | no

</figure><br class="Apple-interchange-newline">

`baa817c` sends the EOS-specific opcode and succeeds:

```
ptp_usb_sendreq   (2): Sending PTP_OC 0x9110 (EOS_SetDevicePropValueEx) request...
ptp_usb_senddata  (2): Sending PTP_OC 0x9110 (EOS_SetDevicePropValueEx) data...
```

`c5a5644` sends the generic one and is refused:

```
_put_Genericu16Table (2): FOUND right value for 100 in the enumeration at val 72
ptp_usb_sendreq      (2): Sending PTP_OC 0x1016 (Set device property value) (0xd103) request...
ptp_usb_senddata     (2): Sending PTP_OC 0x1016 (Set device property value) data...
ptp_usb_getresp [usb.c:516] (0): PTP_OC 0x1016 receiving resp failed: PTP Device Prop Not Supported (0x200a)
_set_config [config.c:12035](0): 'ptp_generic_setdevicepropvalue (params, cursub->propid, &propval, cursub->type)' failed: PTP Device Prop Not Supported (0x200a)
gp_context_error     (0): The property 'ISO Speed' / 0xd103 was not set (0x200a: PTP Device Prop Not Supported)
```

[log-baa817c-WORKS.txt.gz](https://github.com/user-attachments/files/32429081/log-baa817c-WORKS.txt.gz)

[log-c5a5644-FAILS.txt.gz](https://github.com/user-attachments/files/32429082/log-c5a5644-FAILS.txt.gz)

Sija, 2026-09-20:
Most likely fixed by #1310