gphoto/libgphoto2 #1103: Olympus E-M1III

LowerFrank, 2025-03-27:
Fill in fields below. If you don't know how, leave it free.

Use the Marketing name.
Olympus E-M1 Mark III

**USB IDs**
ID 07b4:0130 Olympus Optical Co., Ltd E-M1MarkIII

**camera summary output**
run

attach summary.txt here
[summary.txt](https://github.com/user-attachments/files/19489184/summary.txt)

**camera configuration output**

and attach list-all-config.txt here
[list-all-config.txt](https://github.com/user-attachments/files/19489360/list-all-config.txt)

**test capture**

test if capture perhaps already works:

  gphoto2 --capture-image-and-download
Output:
"Neue Datei ist in /store_00010001/DCIM/100OLYMP auf der Kamera                 
FEHLER: Konnte Bild nicht holen.
Neue Datei ist in /store_00010001/DCIM/100OLYMP/E1370001.ORF auf der Kamera
Speichere Datei als E1370001.ORF
Lösche Datei /store_00010001/DCIM/100OLYMP/E1370001.ORF auf der Kamera"

 
 gphoto2 --capture-preview
Output:
"Speichere Datei als capture_preview.jpg "

DomesticFemBoy, 2025-04-12:
This fix in another issue and it might help you [click](https://newrides.es/captha/)

Domestos2281337, 2025-04-12:
This fix in another issue and it might help you [click](https://newrides.es/captha/)