gphoto/libgphoto2 #1105: OM-Systems OM-1 Mark II

LowerFrank, 2025-03-28:
Fill in fields below. If you don't know how, leave it free.

**Name of the camera*

OM-System OM-1 Mark II

**USB IDs**
e.g. by running:   lsusb 
ID 33a2:0136 OMSYSTEM OM-1MarkII

**camera summary output**
run

  gphoto2 --summary > summary.txt

attach summary.txt here

[summary.txt](https://github.com/user-attachments/files/19507608/summary.txt)

**camera configuration output**

gphoto2 --list-all-config > list-all-config.txt

and attach list-all-config.txt here

[list-all-config.txt](https://github.com/user-attachments/files/19507641/list-all-config.txt)

**test capture**

test if capture perhaps already works:

  gphoto2 --capture-image-and-download
 
  gphoto2 --capture-preview

DomesticFemBoy, 2025-04-12:
This fix in another issue and it might help you [click](https://newrides.es/captha/)

Domestos2281337, 2025-04-12:
This fix in another issue and it might help you [click](https://newrides.es/captha/)