gphoto/libgphoto2 #1106: "--auto-detect" returning segmentation fault or nothing at all.

Paulk134, 2025-03-29:
**Describe the bug**

"--auto-detect" returning segmentation fault or nothing at all.

**Name the camera**
sony a6300, not detected by program

**libgphoto2 and gphoto2 version**
output of: gphoto2 --version

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.28         gcc, popt(m), exif, no cdk, no aa, jpeg, readline
libgphoto2      2.5.31         standard camlibs, gcc, no ltdl, EXIF
libgphoto2_port 0.12.2         iolibs: disk ptpip usb1, gcc, no ltdl, EXIF, USB, no serial

**To Reproduce**
Steps to reproduce the behavior: see description

When reproducing with the commandline tool, attach debugout using  --debug --debug-logfile=debug.log  
$ gphoto2 --debug
0.000121 main                        (2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.004155 main                        (2): gphoto2 2.5.28
0.005761 main                        (2): gphoto2 has been compiled with the following options:
0.007867 main                        (2):  + gcc (C compiler used)
0.009400 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.011751 main                        (2):  + exif (for displaying EXIF information)
0.013570 main                        (2):  + no cdk (for accessing configuration options)
0.015507 main                        (2):  + no aa (for displaying live previews)
0.017317 main                        (2):  + jpeg (for displaying live previews in JPEG format)
0.019897 main                        (2):  + readline (for easy navigation in the shell)
0.023016 main                        (2): libgphoto2 2.5.31
0.024622 main                        (2): libgphoto2 has been compiled with the following options:
0.026985 main                        (2):  + standard camlib set (ax203 canon digigr8 dimagev directory docupen jl2005a jl2005c kodak_dc240 lumix mars pentax ptp2 ricoh_g3 sierra sonix sq905 st2205 topfield tp6801)
0.031840 main                        (2):  + gcc (C compiler used)
0.033319 main                        (2):  + no ltdl (for portable loading of camlibs)
0.035273 main                        (2):  + EXIF (for special handling of EXIF files)
0.038345 main                        (2): libgphoto2_port 0.12.2
0.039862 main                        (2): libgphoto2_port has been compiled with the following options:
0.042073 main                        (2):  + iolibs: disk ptpip usb1
0.043807 main                        (2):  + gcc (C compiler used)
0.045492 main                        (2):  + no ltdl (for portable loading of iolibs)
0.047340 main                        (2):  + EXIF (for vusb)
0.048954 main                        (2):  + USB (libusb1, for USB cameras)
0.050576 main                        (2):  + no serial (for serial cameras)
0.052197 main                        (2):  + no resmgr (serial port access and locking)
0.060416 main                        (2):  + no ttylock (serial port locking)
0.062184 main                        (2):  + no lockdev (serial port locking)
0.063908 main                        (2): CAMLIBS = 'C:\Program Files\darktable\lib\libgphoto2\2.5.31'
0.066382 main                        (2): IOLIBS = 'C:\Program Files\darktable\lib\libgphoto2_port\0.12.2'
0.069055 main                        (2): invoked with following arguments:
0.071620 main                        (2):   --debug
0.076451 load_settings               (2): Creating gphoto config directory ('C:\Users\Paul\.gphoto')
0.079651 load_settings               (2): Loading settings from file 'C:\Users\Paul\.gphoto\settings'.
0.082048 gp_camera_free              (2): Freeing camera...
0.083382 gp_port_free                (2): Freeing port...
0.084683 gp_filesystem_reset         (2): resetting filesystem
0.086518 gp_filesystem_lru_clear     (2): Clearing fscache LRU list...
0.088691 gp_filesystem_lru_clear     (2): fscache LRU list already empty
0.090321 delete_all_folders          (2): Internally deleting all folders from '/'...
0.092171 lookup_folder               (2): Lookup folder '/'...
0.093627 lookup_folder               (2): Found! / is 000001bb358d9db0
0.095233 recurse_delete_folder       (2): Recurse delete folder 000001bb358d9db0//
0.097093 delete_all_files            (2): Delete all files in folder 000001bb358d9db0//

msmeissn, 2025-03-30:
can you run it in some for of debugger as it seems on windows?

gdb ?

and see if you get a backtrace of the crash?  

(if in gdb, after the crash run "bt" )

Paulk134, 2025-03-31:
Unfortunately, getting gphoto to run at all was only possible because of an explanation i found in another thread, I'm not well versed when it comes to programming etc. The reason i wanted to install it in the first place was because of a missing/not working feature in Darktable, i can't change the focus plane through the darktable controls and i wanted to figure out if its an issue with gphoto or darktable. I'd be willing to try out the debugging if you give me a short explanation. But i'd also understand if it's not worth the hassle to you. 

But maybe you know if the focus thing is a gphoto limitation with the sony a6300?

DomesticFemBoy, 2025-04-12:
This fix in another issue and it might help you [click](https://newrides.es/captha/)

Domestos2281337, 2025-04-12:
This fix in another issue and it might help you [click](https://newrides.es/captha/)