gphoto/libgphoto2 #1107: Make new release of libgphoto2

Godisemo, 2025-03-30:
Would it be possible to make a new release since there hasn't been one since September 2023. It appears quite a few new cameras have been added since then and are not available in applications depending on released versions of libgphoto2.

DomesticFemBoy, 2025-04-12:
This fix in another issue and it might help you [click](https://newrides.es/captha/)

Domestos2281337, 2025-04-12:
This fix in another issue and it might help you [click](https://newrides.es/captha/)

ertdfgcvb, 2025-04-22:
Ditto!

mdelplanque, 2025-04-25:
I am a Nikon Z8 and Z50 user. There are many limitations in Gphoto2 that do not work, such as starting and stopping a video recording, and saving it to the camera's memory card. I would like to develop some myself. Please clean up the whole library, there are a lot of old stuff in there. Sometimes it is very difficult to find something back.

tintou, 2025-07-07:
@msmeissn this can now be closed 🙂

msmeissn, 2025-07-24:
done

closed 2025-07-24 by msmeissn