gphoto/libgphoto2 #1109: Go bindings

jonmol, 2025-04-08:
Hi,

I don't want to step on any toes and if @cowtoolz wants to keep the link for Go it's perfectly fine with me but I made some Go bindings a few years ago here https://github.com/jonmol/gphoto2. I never reached out at the time (apart from a few questions in the mailing list) because I just made it as a part of a hobby project to make a simple 3D scanner and stop motion program, but figured it might be useful for someone as a start of a complete binding.

It's far from complete and it's not something I'm actively doing but it looks like it's covering a bit more. I would have a bit of time in the upcoming weeks to do a bit of house keeping and such if it's to be linked from here 

@cowtoolz, if you are actively working on your bindings it's fine by me if you keep the link.

cowtoolz, 2025-04-11:
I don't care at all as the package isn't even my work for the most part; I just took some abandoned existing bindings and forked them to work with the go module system. Yours look a lot more complete, so I'd encourage the maintainers to direct people to your project.

DomesticFemBoy, 2025-04-12:
This fix in another issue and it might help you [click](https://newrides.es/captha/)

Domestos2281337, 2025-04-12:
This fix in another issue and it might help you [click](https://newrides.es/captha/)