gphoto/libgphoto2 #1113: Canon EOS R7 Focus Bracketing, unable trigger or capture and download.

SnibsLabs, 2025-04-23:
**Is your feature request related to a problem? Please describe.**
I can not use --trigger-capture or --set-config bracketmode=1. 

Setting the Canon R7 with Focus Bracketing + Depth Composite options as "Enable" lets the camera to be triggered, but it freezes on composing the final picture and becomes unresponsive. 

--set-config bracketmode=1 results in an error, 

**Describe the solution you'd like**
The goal is to have a in-camera Focus Bracketing, Depth composite picture triggered with gphoto2 and preferably downloaded onto a computer immediately, with a loop that would allow for hundreds of consecutive excecutions of this. 

**Describe alternatives you've considered**
Connecting through Canons own utility tool, I am able to trigger one of these images and get it onto the computer. EDSDK is another route, but I would like to figure out a way to handle this through gphoto2. I tried to undestand what would it require to dig out this information from the USB communications but as of now I dont have a clue on how to proceed. The setting is turned on in the camera, but listed as Unkown value 0000. I also was concidering is there a way to trigger EOS commands through the gphoto2, but I was unable to find any material on it.

/main/capturesettings/bracketmode
Label: Bracket Mode
Readonly: 0
Type: RADIO
Current: Unknown value 0000
Choice: 0 Unknown value 0000
END

**Additional context**
add bracketmode.log and trigger.log. Paste gphoto2 --summary

gphoto2 --summary

*** Error ***
error 0xa106

*** Error ***
error 0xa106
Camera summary:
Manufacturer: Canon.Inc
Model: Canon EOS R7
  Version: 3-1.3.2
  Serial Number: removed
Vendor Extension ID: 0xb (1.0)
Vendor Extension Description:

Capture Formats: JPEG Unknown(b108) Unknown(b10b) Unknown(b982)
Display Formats: Association/Directory, Script, DPOF, MS AVI, MS Wave, JPEG, Unknown(b103), Unknown(bf02), Defined Type, Unknown(b105), Unknown(b982), Unknown(b10a), Unknown(b10b), Unknown(b109)

Device Capabilities:
        File Download, File Deletion, File Upload
        No Image Capture, No Open Capture, Canon EOS Capture, Canon EOS Capture 2
        Canon Wifi support

Storage Devices Summary:
store_00010001:
        StorageDescription: SD1
        VolumeLabel:
        Storage Type: Removable RAM (memory card)
        Filesystemtype: Digital Camera Layout (DCIM)
        Access Capability: Read-Write
        Maximum Capability: 7939817472 (7572 MB)
        Free Space (Bytes): 7011270656 (6686 MB)
        Free Space (Images): -1

Device Property Summary:
Property 0xd402:(read only) (type=0xffff) 'Canon EOS R7'
Property 0xd407:(read only) (type=0x6) 1
Property 0xd406:(readwrite) (type=0xffff) 'Windows/10.0.26100 MTPClassDriver/10.0.26100.0'
Property 0xd303:(read only) (type=0x2) 1
Battery Level(0x5001):(read only) (type=0x2) Enumeration [100,0,75,0,50] value: 73% (73)

[image]

[bracketmode.log](https://github.com/user-attachments/files/19867164/bracketmode.log)

[trigger.log](https://github.com/user-attachments/files/19867160/trigger.log)

SnibsLabs, 2025-04-23:
gphoto2 --get-config bracketmode
Label: Bracket Mode
Readonly: 0
Type: RADIO
Current: Unknown value 0000
Choice: 0 Unknown value 0000
END

RafatH0ssain, 2026-10-29:
I looked into this on my R7 (firmware 1.8.0). Focus bracketing doesn't go through bracketmode (0xD11D) at all. It's 0xD208, which ptp.h already names EOS_FocusShiftSetting, but config.c never exposes it for Canon. The 0xD208 entries in config.c are all the Fuji camera action. On my R7 bracketmode reads 0 with only 0 offered, whether focus bracketing is on or off, so setting bracketmode=1 can't work.

0xD208 is a 36-byte value, nine 32-bit fields. I mapped these by changing settings in the menu and reading the EOS property dump:

0x00  36   size
0x04  3    ?
0x08  0/1  focus bracketing off/on
0x0C  50   number of shots (set 10, read 10)
0x10  3    focus increment (set 7, read 7)
0x14  1    ? (probably exposure smoothing)
0x18  0/1  depth compositing off/on
0x1C  1    ? (probably crop depth composite, reads -1 when compositing is off)
0x20  -2   ?

Also, I couldn't reproduce the freeze on 1.8.0, so it may be fixed in firmware after the 1.3.2 in your summary. With focus bracketing and depth compositing on, trigger_capture worked and the camera stayed responsive. It only made a composite with capturetarget set to Memory card, though. With the default Internal RAM, the burst stopped part way and no composite was made. Also, with the card target gphoto2 only reported the composite as a new file, not the source frames, even though they're all on the card.

I can work on a PR that exposes focus bracketing on/off, shots and increment as config entries, and test it on the R7. Let me know if that would be the right approach!

vindavs, 2026-10-31:
@RafatH0ssain thanks for the mapping. I ran into the same thing on an R6 Mark II and ended up with a working version, now in #1341. If you could try it on your R7, that would be great.