gphoto/libgphoto2 #1129: Unable to recover Sony ZV-E10 USB connection after power interruption

JavierDroneit, 2025-06-06:
I'm using a Sony ZV-E10 connected to a Raspberry Pi for an automated photo capture setup. The camera is powered via a dedicated external battery (not through USB), and the USB connection is used solely for data transfer.

Under normal conditions, everything works perfectly. The camera is correctly detected and works with gphoto2.

```
gphoto2 --auto-detect
Model                          Port                                            
----------------------------------------------------------
Sony ZV-E10 (Control)          usb:001,007  
```

Occasionally, due to power issues or reboots, the camera loses power. After such events, once power is restored, the USB connection no longer works as expected. Running **lsusb** shows:

> Sony Corp. ILCE-6000 (aka Alpha-6000) in charging mode

USB power charging is disabled in the camera settings.
I have tested this same setup with multiple other camera models, and none of them exhibit this behavior.

In this state, I cannot communicate with the camera. I have tried using **usbreset** and **uhubctl** to reset the USB port, but the camera remains stuck in this mode. The only way to recover communication is to physically toggle the camera’s ON/OFF switch. This breaks the intended automation flow, since physical interaction defeats the purpose.

I found a similar issue possibly related with a different camera model: [#343](https://github.com/gphoto/libgphoto2/issues/343)

Software versions:

gphoto2: 2.5.28
libgphoto2: 2.5.30

knoppmyth, 2026-03-01:
This is interesting as several years back, I was on a project and we used the DSC-QX10 via the REST API. When it worked, it was great. However, we started to run into this exact same issue. In order to get the camera working again, we had to physically turn it off. Something, we physically had to remove the battery to get it functioning again. 

One year at CES, I spoke w/ one of the Sony engineers and he said it sounded like a power issue. His suggestion was trying to ensure the power to the camera was "clean". I tried using a power bank but that wasn't any better... 

Pity this issue is apparently still around. I was consider a ZV-E10 II.

jufa, 2026-03-01:
Another possibility is the usb c cable, have you tried a few they have internal logic that can vary in behaviour. 

My latest similar experience using zve10s (mk I) for automation with pc remote and dummy battery and frequent power outages (Linux):

I set the camera, using its rear controls, to the configuration I want, then save this to memory location 1 (internal), then set the camera to use this MR1 (the same memory slot). Part of these setting is the USB connection mode in tab 3 of the menu system.  

A trick I found with storing settings is that after doing the above procedure, it is not stored properly unless I power off the zve10 using its power switch after storing  new settings and wait may be 15 to 30 seconds. This is only needed when changing the stores settings. 

That way after a power interruption, it should boot up to MR1 and associated USB config. Please let us know if this works for you.

knoppmyth, 2026-03-02:
@jufa Please clarify, you've not seen this issue w/ your ZV-EV10 MkI? Your advice would be, after adjusting your settings, save it and turn the camera off to ensure the settings are indeed saved and to use a quality USB-C cable?