gphoto/libgphoto2 #1135: PTP/IP connection to Nikon cameras (e.g., Z 8) fails to query properties, results in "PTP error 2005"

INI-K, 2025-07-07:
I am having a consistent issue connecting to my Nikon cameras via PTP/IP. It appears that all of my Nikon models behave the same way. While a basic connection is established, gphoto2 is unable to query almost any device properties and reports very limited capabilities.

Could you please help me understand why this is happening and if there is a potential solution?

0.000002 main                        (2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.000008 main                        (2): gphoto2 2.5.28.1
0.000010 main                        (2): gphoto2 has been compiled with the following options:
0.000011 main                        (2):  + gcc (C compiler used)
0.000012 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.000013 main                        (2):  + exif (for displaying EXIF information)
0.000014 main                        (2):  + no cdk (for accessing configuration options)
0.000015 main                        (2):  + no aa (for displaying live previews)
0.000015 main                        (2):  + no jpeg (for displaying live previews in JPEG format)
0.000016 main                        (2):  + no readline (for easy navigation in the shell)
0.000039 main                        (2): libgphoto2 2.5.32.1
0.000051 main                        (2): libgphoto2 has been compiled with the following options:
0.000054 main                        (2):  + standard camlib set SKIPPING SOME (ax203 canon digigr8 dimagev directory jl2005a jl2005c kodak_dc240 lumix mars pentax ptp2 quicktake1x0 ricoh_g3 sierra sonix sq905 st2205 topfield tp6801 SKIPPING docupen)
0.000056 main                        (2):  + gcc (C compiler used)
0.000058 main                        (2):  + no ltdl (for portable loading of camlibs)
0.000059 main                        (2):  + EXIF (for special handling of EXIF files)
0.000060 main                        (2): libgphoto2_port 0.12.2
0.000062 main                        (2): libgphoto2_port has been compiled with the following options:
0.000063 main                        (2):  + iolibs: disk ptpip serial usb1
0.000063 main                        (2):  + gcc (C compiler used)
0.000065 main                        (2):  + no ltdl (for portable loading of iolibs)
0.000065 main                        (2):  + EXIF (for vusb)
0.000066 main                        (2):  + USB (libusb1, for USB cameras)
0.000067 main                        (2):  + serial (for serial cameras)
0.000068 main                        (2):  + no resmgr (serial port access and locking)
0.000069 main                        (2):  + no ttylock (serial port locking)
0.000070 main                        (2):  + no lockdev (serial port locking)
0.000071 main                        (2): CAMLIBS env var not set, using compile-time default instead
0.000072 main                        (2): IOLIBS env var not set, using compile-time default instead
0.000072 main                        (2): invoked with following arguments:
0.000074 main                        (2):   --debug
0.000075 main                        (2):   --port=ptpip:192.168.147.75
0.000076 main                        (2):   --summary
0.000135 main                        (2): Processing 'port' option ('ptpip:192.168.147.75')...
0.000344 gp_port_init_localedir      (2): localedir has been set to /usr/local/share/locale (compile-time default)
0.000348 gp_port_info_list_load      (2): Using ltdl to load io-drivers from '/usr/local/lib/libgphoto2_port/0.12.2'...
0.000428 foreach_func                (2): Called for filename '/usr/local/lib/libgphoto2_port/0.12.2/disk'.
0.001265 foreach_func                (2): Loaded '' ('^disk:') from '/usr/local/lib/libgphoto2_port/0.12.2/disk'.
0.001268 foreach_func                (2): Called for filename '/usr/local/lib/libgphoto2_port/0.12.2/ptpip'.
0.002720 foreach_func                (2): Loaded 'PTP/IP Connection' ('ptpip:') from '/usr/local/lib/libgphoto2_port/0.12.2/ptpip'.
0.002723 foreach_func                (2): Loaded '' ('^ptpip:') from '/usr/local/lib/libgphoto2_port/0.12.2/ptpip'.
0.002724 foreach_func                (2): Loaded 'IP Connection' ('ip:') from '/usr/local/lib/libgphoto2_port/0.12.2/ptpip'.
0.002725 foreach_func                (2): Loaded '' ('^ip:') from '/usr/local/lib/libgphoto2_port/0.12.2/ptpip'.
0.002726 foreach_func                (2): Called for filename '/usr/local/lib/libgphoto2_port/0.12.2/serial'.
0.005574 foreach_func                (2): Loaded 'Serial Port Device' ('serial:') from '/usr/local/lib/libgphoto2_port/0.12.2/serial'.
0.005581 foreach_func                (2): Loaded '' ('^serial:') from '/usr/local/lib/libgphoto2_port/0.12.2/serial'.
0.005582 foreach_func                (2): Called for filename '/usr/local/lib/libgphoto2_port/0.12.2/usb1'.
0.010948 foreach_func                (2): Loaded '' ('^usb:') from '/usr/local/lib/libgphoto2_port/0.12.2/usb1'.
0.010954 foreach_func                (2): Loaded 'Universal Serial Bus' ('usb:000,007') from '/usr/local/lib/libgphoto2_port/0.12.2/usb1'.
0.010955 foreach_func                (2): Loaded 'Universal Serial Bus' ('usb:000,006') from '/usr/local/lib/libgphoto2_port/0.12.2/usb1'.
0.010956 foreach_func                (2): Loaded 'Universal Serial Bus' ('usb:000,005') from '/usr/local/lib/libgphoto2_port/0.12.2/usb1'.
0.010958 foreach_func                (2): Loaded 'Universal Serial Bus' ('usb:000,003') from '/usr/local/lib/libgphoto2_port/0.12.2/usb1'.
0.010959 foreach_func                (2): Loaded 'Universal Serial Bus' ('usb:003,001') from '/usr/local/lib/libgphoto2_port/0.12.2/usb1'.
0.010961 gp_port_info_list_count     (2): Counting entries (13 available)...
0.010963 gp_port_info_list_count     (2): 8 regular entries available.
0.010964 gp_port_info_list_lookup_path(2): Looking for path 'ptpip:192.168.147.75' (13 entries available)...
0.010965 gp_port_info_list_lookup_path(2): Starting regex search for 'ptpip:192.168.147.75'...
0.010966 gp_port_info_list_lookup_path(2): Trying '^disk:'...
0.010979 gp_port_info_list_lookup_path(2): regexec failed
0.010980 gp_port_info_list_lookup_path(2): Trying '^ptpip:'...
0.010989 gp_port_info_list_get_info  (2): Getting info of entry 8 (14 available)...
0.010991 gp_camera_set_port_info     (2): Setting port info for port 'Generic Port' at 'ptpip:192.168.147.75'...
0.011330 gp_port_set_settings        (2): Setting settings...
0.011334 load_settings               (2): Creating gphoto config directory ('/Users/meo/.gphoto')
0.011492 load_settings               (2): Loading settings from file '/Users/meo/.gphoto/settings'.
0.011508 gp_setting_set              (2): Setting key 'port' to value 'ptpip:192.168.147.75' (gphoto2)
[... 454 lines cut ...]
0.736990 gp_port_close               (2): Closing port...
0.737186 gp_filesystem_reset         (2): resetting filesystem
0.737191 gp_filesystem_lru_clear     (2): Clearing fscache LRU list...
0.737193 gp_filesystem_lru_clear     (2): fscache LRU list already empty
0.737195 delete_all_folders          (2): Internally deleting all folders from '/'...
0.737196 lookup_folder               (2): Lookup folder '/'...
0.737197 lookup_folder               (2): Found! / is 0x600002fb4030
0.737198 recurse_delete_folder       (2): Recurse delete folder 0x600002fb4030//
0.737199 delete_all_files            (2): Delete all files in folder 0x600002fb4030//
0.737200 gp_port_free                (2): Freeing port...
0.737202 gp_port_close               (2): Closing port...
0.737326 gp_filesystem_reset         (2): resetting filesystem
0.737330 gp_filesystem_lru_clear     (2): Clearing fscache LRU list...
0.737330 gp_filesystem_lru_clear     (2): fscache LRU list already empty
0.737331 delete_all_folders          (2): Internally deleting all folders from '/'...
0.737332 lookup_folder               (2): Lookup folder '/'...
0.737333 lookup_folder               (2): Found! / is 0x600002fb4030
0.737334 recurse_delete_folder       (2): Recurse delete folder 0x600002fb4030//
0.737335 delete_all_files            (2): Delete all files in folder 0x600002fb4030//
**To Reproduce**
Steps to reproduce the behavior:

When reproducing with the commandline tool, attach debugout using  --debug --debug-logfile=debug.log   if considered useful.

```

msmeissn, 2025-07-08:
According to the debug trace it does not offer any device property management opcodes in STA Mode, its really just download options in the log.

INI-K, 2025-07-09:
@msmeissn 
I think I found a reason.
But I couldn't find this anywhere.

I found out that the process of authenticating with the operation code is different when connecting in STA mode and AP mode in Nikon.

Currently, the connection is successful in STA mode.
Problem is I'm looking for a way to run gPhoto2 in the same session.

INI-K, 2025-07-15:
@msmeissn 

I wonder if you are willing to show the API that transmits the PTPIP operation code to the outside.

I can show you the codes I've succeeded in.

Even in AP mode
The Nikon camera is not functioning normally.
The Nikon camera is in smart device mode
There's a PC mode

I don't think the difference between the two modes is very different
The process of authenticating is a little different.
Currently, gphoto2's ptpip connection is not maintained, and only the result is cut off after sending it right away. (Camera ends the connection.)
This seems to require the process of maintaining the connection (do not close the session)