gphoto/libgphoto2 #1153: 2.5.32 breaks (or changes)  Canon EOS  format and ISO functionality

MurrayVK3, 2025-08-14:
I upgraded to libgphoto 2.5.32 and can no longer set the image format or ISO on a Canon EOS 550D via Ekos and Indi.

The image format choices change with the later version:

With libgphoto 2.5.30 

>gphoto2 --get-config  /main/imgsettings/imageformat
> 		Label: Image Format                                                            
> 		Readonly: 0
> 		Type: RADIO
> 		Current: Small Normal JPEG
> 		Choice: 0 Large Fine JPEG
> 		Choice: 1 Large Normal JPEG
> 		Choice: 2 Medium Fine JPEG
> 		Choice: 3 Medium Normal JPEG
> 		Choice: 4 Small Fine JPEG
> 		Choice: 5 Small Normal JPEG
> 		Choice: 6 RAW + Large Fine JPEG
> 		Choice: 7 RAW
> 		END
> 

With libgphoto 2.5.32

>gphoto2 --get-config  /main/imgsettings/imageformat
> 		Label: Image Format                                                            
> 		Readonly: 0
> 		Type: RADIO
> 		Current: cS
> 		Choice: 0 L
> 		Choice: 1 cL
> 		Choice: 2 M
> 		Choice: 3 cM
> 		Choice: 4 S
> 		Choice: 5 cS
> 		Choice: 6 RAW + L
> 		Choice: 7 RAW
> 		END
> 

The format menu in Ekos gets populated from the above but with version 2.5.32 it no longer changes the setting in the camera.
Also with version 2.5.32 the ISO menu although correctly populated has no effect on the camera setting.

Reverting to libgphoto version 2.5.30 fixes both the above issues.

Testing with gphoto2 I can successfully set both the format and ISO with either versions of libgphoto2 which suggests the issue may be with Indi or Ekos but upgrading/downgrading either or both those makes no difference. The only way to fix it is to revert libgphoto2 back to 2.5.30.  At that point it doesn't make any difference which version of Indi or Ekos I use, only the version of libgphoto2 matters.

**Name the camera**

> gphoto2 --auto-detect 
>         Model                          Port                                            
>         ----------------------------------------------------------
>         Canon EOS 550D                 usb:002,027  
>         

**With format and ISO working:**

> gphoto2 --version
> gphoto2 2.5.28
> 		
> This version of gphoto2 is using the following software versions and options:
> gphoto2         2.5.28         gcc, popt(m), exif, cdk, aa, jpeg, readline
> libgphoto2      2.5.30         standard camlibs, gcc, no ltdl, EXIF
> libgphoto2_port 0.12.1         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, no ltdl, EXIF, USB, serial without locking
> 

**Format and ISO not working:**
> gphoto2 --version
> gphoto2 2.5.32.1
> 
> This version of gphoto2 is using the following software versions and options:
> gphoto2         2.5.32.1       gcc, popt(m), exif, no cdk, no aa, jpeg, readline
> libgphoto2      2.5.32.1       standard camlibs (SKIPPING docupen), gcc, no ltdl, EXIF
> libgphoto2_port 0.12.2         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, no ltdl, EXIF, USB, serial without locking
>

axxel, 2025-09-28:
Sorry that my contribution in #1031 caused your troubles. I anticipated as much but please see [this comment](https://github.com/gphoto/libgphoto2/pull/1031#issuecomment-2381600594) for a detailed explanation why I chose to propose this change. Please report this to the Indi/Ekos people for them to make sure they play along with the new names.

Sija, 2026-09-20:
Most likely fixed by #1310

mnigbur, 2026-09-24:
Can confirm that fixed it. Tested with 600D & 1300D.

```
gphoto2 --debug --debug-logfile=log-<v>.txt \
  --set-config capture=on \
  --set-config iso=100 --set-config iso=6400 \
  --set-config imageformat=cL --set-config imageformat=RAW
```

buggy (5672d5104): PTP traffic for the ISO Speed (0xd103) set
```
0.265638 _set_config                 (2): Setting property 'ISO Speed' / 0xd103
0.265640 _put_Genericu16Table        (2): FOUND right value for 100 in the enumeration at val 0048
0.265643 ptp_usb_sendreq             (2): Sending PTP_OC 0x1016 (Set device property value) (0xd103) request...
0.265644 gp_port_write               (3): Writing 16 = 0x10 bytes to port...
0.265728 gp_port_write               (3): Wrote   16 = 0x10 bytes to port: (hexdump of 16 bytes)
0000  10 00 00 00 01 00 16 10-2f 00 00 00 03 d1 00 00  ......../.......

0.265730 ptp_usb_senddata            (2): Sending PTP_OC 0x1016 (Set device property value) data...
0.265732 gp_port_write               (3): Writing 14 = 0xe bytes to port...
0.265768 gp_port_write               (3): Wrote   14 = 0xe bytes to port: (hexdump of 14 bytes)
0000  0e 00 00 00 02 00 16 10-2f 00 00 00 48 00        ......../...H.  

0.265770 ptp_usb_getresp             (2): Reading PTP_OC 0x1016 (Set device property value) response...
0.265772 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...
0.267111 gp_port_read                (3): Read    12 = 0xc out of 1024 bytes from port: (hexdump of 12 bytes)
0000  0c 00 00 00 03 00 0a 20-2f 00 00 00              ....... /...    

0.267117 ptp_usb_getresp [usb.c:495] (0): PTP_OC 0x1016 receiving resp failed: PTP Device Prop Not Supported (0x200a)
0.267119 _set_config [config.c:13150](0): 'ptp_generic_setdevicepropvalue (params, cursub->propid, &propval, cursub->type)' failed: PTP Device Prop Not Supported (0x200a)
0.267124 gp_context_error            (0): The property 'ISO Speed' / 0xd103 was not set (0x200a: PTP Device Prop Not Supported)
0.267162 gp_context_error            (0): Property 'iso' not found.
0.267168 gp_camera_set_single_config [gphoto2-camera.c:1141](0): 'camera->functions->set_single_config ( camera, name, widget, context)' failed: -2
0.267172 gp_context_error            (0): Failed to set new configuration value 100 for configuration entry iso.
0.267180 ptp_usb_sendreq             (2): Sending PTP_OC 0x9116 (EOS_GetEvent) request...
0.267182 gp_port_write               (3): Writing 12 = 0xc bytes to port...
0.267241 gp_port_write               (3): Wrote   12 = 0xc bytes to port: (hexdump of 12 bytes)
0000  0c 00 00 00 01 00 16 91-30 00 00 00              ........0...    
```

```
0.275249 _set_config                 (2): Setting property 'Image Format' / 0xd120
0.275259 _put_Canon_EOS_ImageFormat  (2): FOUND right value for cL in the enumeration at val 02ff
0.275261 ptp_usb_sendreq             (2): Sending PTP_OC 0x1016 (Set device property value) (0xd120) request...
0.275263 gp_port_write               (3): Writing 16 = 0x10 bytes to port...
0.275362 gp_port_write               (3): Wrote   16 = 0x10 bytes to port: (hexdump of 16 bytes)
0000  10 00 00 00 01 00 16 10-35 00 00 00 20 d1 00 00  ........5... ...

0.275364 ptp_usb_senddata            (2): Sending PTP_OC 0x1016 (Set device property value) data...
0.275366 gp_port_write               (3): Writing 14 = 0xe bytes to port...
0.275401 gp_port_write               (3): Wrote   14 = 0xe bytes to port: (hexdump of 14 bytes)
0000  0e 00 00 00 02 00 16 10-35 00 00 00 ff 02        ........5.....  

0.275404 ptp_usb_getresp             (2): Reading PTP_OC 0x1016 (Set device property value) response...
0.275405 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...
0.276748 gp_port_read                (3): Read    12 = 0xc out of 1024 bytes from port: (hexdump of 12 bytes)
0000  0c 00 00 00 03 00 0a 20-35 00 00 00              ....... 5...    

0.276751 ptp_usb_getresp [usb.c:495] (0): PTP_OC 0x1016 receiving resp failed: PTP Device Prop Not Supported (0x200a)
0.276753 _set_config [config.c:13150](0): 'ptp_generic_setdevicepropvalue (params, cursub->propid, &propval, cursub->type)' failed: PTP Device Prop Not Supported (0x200a)
0.276756 gp_context_error            (0): The property 'Image Format' / 0xd120 was not set (0x200a: PTP Device Prop Not Supported)
0.276781 gp_context_error            (0): Property 'imageformat' not found.
0.276791 gp_camera_set_single_config [gphoto2-camera.c:1141](0): 'camera->functions->set_single_config ( camera, name, widget, context)' failed: -2
0.276794 gp_context_error            (0): Failed to set new configuration value cL for configuration entry imageformat.
0.276801 ptp_usb_sendreq             (2): Sending PTP_OC 0x9116 (EOS_GetEvent) request...
0.276816 gp_port_write               (3): Writing 12 = 0xc bytes to port...
0.276871 gp_port_write               (3): Wrote   12 = 0xc bytes to port: (hexdump of 12 bytes)
0000  0c 00 00 00 01 00 16 91-36 00 00 00              ........6...    
```

[... 105 lines cut ...]

0.354595 ptp_usb_getresp             (2): Reading PTP_OC 0x9116 (EOS_GetEvent) response...
0.354596 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...
0.355039 gp_port_read                (3): Read    12 = 0xc out of 1024 bytes from port: (hexdump of 12 bytes)
0000  0c 00 00 00 03 00 01 20-39 00 00 00              ....... 9...    

0.355054 _get_config                 (2): Getting property 'Image Format' / 0xd120
[... cut ...]
0000  14 00 00 00 02 00 16 91-3a 00 00 00 08 00 00 00  ........:.......
0010  00 00 00 00            -                         ....            

0.356221 ptp_usb_getresp             (2): Reading PTP_OC 0x9116 (EOS_GetEvent) response...
0.356225 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...
0.356658 gp_port_read                (3): Read    12 = 0xc out of 1024 bytes from port: (hexdump of 12 bytes)
0000  0c 00 00 00 03 00 01 20-3a 00 00 00              ....... :...    
```

closed 2026-09-25 by msmeissn