gphoto/libgphoto2 #1156: OM-1 Mark II fails to capture and download first image

coldfire22x, 2025-08-22:
When the memory card is empty, the first `gphoto2 --capture-image-and-download` will take and record an image to the SD card but fail to download it. All subsequent attempts will capture and download the image, then remove it from the SD card.

E.g., if `gphoto2 --capture-image-and-download` is called 5 times in succession, the 1st image will be recorded to the SD card but not downloaded. The next 4, however, will be downloaded and removed from the SD card.

**Name the camera**
```
Model                          Port
----------------------------------------------------------
Olympus OM-1MarkII             usb:005,015
```

**libgphoto2 and gphoto2 version**
```
gphoto2         2.5.32         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.32         standard camlibs, gcc, no ltdl, EXIF
libgphoto2_port 0.12.2         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, no ltdl, EXIF, USB, serial without locking
```

**To Reproduce**
First delete all images from the SD card.

```
$ gphoto2 --capture-image-and-download --debug --debug-loglevel=debug --debug-logfile=1-first-attempt.log
ERROR: Could not capture image.
ERROR: Could not capture.

$ gphoto2 --capture-image-and-download --debug --debug-loglevel=debug --debug-logfile=2-second-attempt.log
New file is in location /store_00010001/DCIM/100OMSYS/P8210001.ORF on the camera
Saving file as P8210001.ORF
Deleting file /store_00010001/DCIM/100OMSYS/P8210001.ORF on the camera
```

[1-first-attempt.log](https://github.com/user-attachments/files/21928040/1-first-attempt.log)
[2-second-attempt.log](https://github.com/user-attachments/files/21928041/2-second-attempt.log)

[summary.log](https://github.com/user-attachments/files/21928154/summary.log)

mszyndel, 2026-10-31:
I just retested this. Still doesn't work, but I got a slightly different result that you did. Single card, formatted prior to the test.

First file is indeed on the SD card, so capture worked, just download failed.

```
> bin/gphoto2" --capture-image-and-download   --debug --debug-loglevel=debug --debug-logfile=first-capture.log
New file is in location /store_00010001/DCIM on the camera                     
ERROR: Could not get image.
New file is in location /store_00010001/DCIM/100OMSYS on the camera
ERROR: Could not get image.
> bin/gphoto2" --capture-image-and-download   --debug --debug-loglevel=debug --debug-logfile=second-capture.log
New file is in location /store_00010001/DCIM/100OMSYS/P1010002.ORF on the camera
Saving file as P1010002.ORF
Deleting file /store_00010001/DCIM/100OMSYS/P1010002.ORF on the camera
```

[first-capture.log](https://github.com/user-attachments/files/34216248/first-capture.log)
[second-capture.log](https://github.com/user-attachments/files/34216249/second-capture.log)

coldfire22x, 2026-10-31:
I haven't tested this one in a while .. I usually just make sure I manually take the first photo before imaging with libgphoto2.  If I get some time this weekend, I'll try to take another look at it.

mszyndel, 2026-10-31:
Actually there was already a TODO comment mentioning the root cause of this issue. Fixed in https://github.com/gphoto/libgphoto2/pull/1342

closed 2026-11-01 by msmeissn