gphoto/libgphoto2 #1161: OM-1 Mark II fails to capture in MF

coldfire22x, 2025-08-24:
I think this is actually the issue I was experiencing in #969.

When in S-AF mode, capture and download works (as long as the card isn't empty, see #1156). However, when set to MF, it fails unless the line mentioned [here](https://github.com/gphoto/libgphoto2/issues/969#issuecomment-2080430962) is commented out.

For what it's worth, it doesn't seem like the OM-1 Mark II likes setting `0xd052` to anything other than `2`, compared to the E-M1 models.  Is it possible this command just needs skipped on the new OM-# bodies?

I'll play more later, but if anyone could shed more insight on what `0xd052` is and how they think it's supposed to work, that'd be much appreciated!

**Name the camera**
```
Model                          Port
----------------------------------------------------------
Olympus OM-1MarkII             usb:005,031
```

**libgphoto2 and gphoto2 version**
```
gphoto2         2.5.32         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.32         standard camlibs, gcc, no ltdl, EXIF
libgphoto2_port 0.12.2         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, no ltdl, EXIF, USB, serial without locking
```

**To Reproduce**
```
$ gphoto2 --capture-image-and-download --debug --debug-loglevel=debug --debug-logfile=S-AF.txt
New file is in location /store_00010001/DCIM/100OMSYS/P8240001.JPG on the camera
Saving file as P8240001.JPG
Deleting file /store_00010001/DCIM/100OMSYS/P8240001.JPG on the camera

$ gphoto2 --capture-image-and-download --debug --debug-loglevel=debug --debug-logfile=MF.txt
ERROR: Could not capture image.
ERROR: Could not capture.
```

[summary.txt](https://github.com/user-attachments/files/21959284/summary.txt)
[list-all-config.txt](https://github.com/user-attachments/files/21959286/list-all-config.txt)
[S-AF.txt](https://github.com/user-attachments/files/21959287/S-AF.txt)
[MF.txt](https://github.com/user-attachments/files/21959288/MF.txt)

mszyndel, 2026-10-31:
@coldfire22x I retested this on latest master and I believe this issue is fixed. Tested on OM-1 mk I with focus set to MF in body.

```
"bin/gphoto2" --capture-image-and-download --debug --debug-loglevel=debug --debug-logfile=mf-test.log
```

[mf-test.log](https://github.com/user-attachments/files/34216120/mf-test.log)

coldfire22x, 2026-10-31:
Can confirm, this issue was resolved by changes that are now in master

closed 2026-10-31 by coldfire22x